FRC2706/2019-2706-Robot-Code #1: Get GradleRIO set up and running

KyleRAnderson, 2018-10-25:
It's a good idea to get used to the new robot framework before the actual kickoff date, so it'd be a good idea to start using it and seeing what it's like. Observations and issues found with the new framework should be documented.

KyleRAnderson, 2018-11-07:
So we've made instructions for cloning our GradleRIO repo with VS code as well as IntelliJ IDEA, currently making a guide for setting it up on Eclipse.

greyingjay, 2018-11-27:
Where is the guide to using VS Code?

KyleRAnderson, 2018-11-27:
We've just got a quick set-up guide mentioned in the [README](https://github.com/FRC2706/2019-2706-Robot-Code/blob/c970daf1eaec948f1497569ca6ab17f43663dd4d/README.md) in the branch `2019-gradlerio` currently not merged to master yet since the code has not been tested (by deploying).

robertlucyshyn, 2018-11-27:
Is there a way to merge only the README to master so everyone can see it? I'm sure others joining will want to refer to something, even if it has not been completely tested. You can put a disclaimer at the top that it still needs to be tested.

greyingjay, 2018-11-27:
I'm bringing Jules tomorrow, as well as the standalone RoboRio.
Between those, and Plyboy, hopefully soon we can try it out.

KyleRAnderson, 2018-11-27:
We could theoretically just merge the README over @robertlucyshyn however the README applies only to the code in that branch.

I'd rather not merge anything untested to master since that would kind of defeat the purpose of the master branch. We're probably going to test next meeting and get the code merged over, at which point we'll be able to get the rookies set up.

robertlucyshyn, 2018-11-27:
Yes, I understand. I'd normally not consider merging anything to master without testing---merging code would be a definite no. I just wanted to make the VSCode setup instructions available to newcomers so they could get set up. Given that we'll likely test this at the next meeting, this is no longer an issue. On the subject and testing and procedures, would it be possible for someone to document how we test code with the different hardware components? I will unfortunately not be at the next meeting and those procedures would be of interest to many people in the future. Thanks.

KyleRAnderson, 2018-11-28:
Good idea to document or give a quick start guide on how to test. Issue has been made: #16.

greyingjay, 2018-11-28:
On a related note, I've been keeping track of my experiences trying to set up a new RoboRio and a new laptop with no tools installed. See here: https://github.com/FRC2706/KnowledgeBase/wiki/Setting-Up-For-2019-Java-Software-Development

greyingjay, 2018-11-29:
I was able to, finally, deploy from scratch to Jules this evening. I think the issue was the RoboRio was not on the latest image. I updated it from v16 to v17 and I was able to deploy basic code that activated the climber motor (driven by a Talon SRX on CAN). My next step will be to ensure that I can get NavX code working, and work through any gradle issues accordingly.

My understanding is that Ryan and Kyle made similar progress on Mergio.

KyleRAnderson, 2018-12-06:
Completed this in #17

closed 2018-12-06 by KyleRAnderson