FRC2706/2019-2706-Robot-Code #3: Document our code processes

KyleRAnderson, 2018-10-25:
It's a good idea for us to document how we do things in software as well as the logic behind a lot of our code and stuff. This is very important this year in particular because we've got a lot of rookies who would benefit from knowing how the code works as well as a rookie team that we're attempting to mentor.

I was thinking of stuff like maybe explaining autonomous math in detail in the repository's wiki pages or something to that effect. As long as it's easily accessible.

**Things to include in documentation**:
- [] How we got the repo set up (with gradle and such) and how we're going to structure it.
- [] How autonomous works (with math explanations a little bit maybe?)

greyingjay, 2018-10-25:
Yes. As "step by step" would be good. I'd like to see this approach taken everywhere we do stuff. (For example, how do I set up a Raspberry Pi to display videos for the Tower of Power?)

We can either use the wiki section of each repo or use our KnowledgeBase repo (which currently doesn't have much) for general info.

KyleRAnderson, 2018-10-25:
Sounds like a good idea. I guess we could put anything general in the KnowledgeBase repo and then anything specific to this year (or this year's repo at least) in this year's wiki.

eandr127, 2019-02-16:
I've added some more wiki pages that should provide all the information listed here.

closed 2019-02-16 by eandr127