FRC2706/2019-2706-Robot-Code #8: Figure out Robot Configuration Manager

KyleRAnderson, 2018-11-15:
We need to have some sort of method for easily accessing robot configurations, such as the button mappings for controllers as well as robot-dependent variables. 

This is kind of like a merge between RobotConfig and RobotMap from last year.

The frontend portion of this needs to be easy to use.

eandr127, 2018-11-17:
I rewrote `RobotMap` in 1bca176f7ea32a6b0225f7b60db188b93ce01e75, but I'm still not sure how to combine all of the different configuration systems into one. Do you have any ideas?

KyleRAnderson, 2018-11-17:
I think it's something we need to discuss for sure because there are many ways to do it.

One way would be to use a JSON file to store the configurations and then just read the correct configuration. We could have "global" configurations for stuff that is not specific to a robot in particular (such as controller mappings) and then "local" configurations for robot-dependent settings.

I'll try and quickly write something to show you what I mean, but since this is going to be used a lot I feel like it's important that we get it right.

KyleRAnderson, 2018-11-18:
It would be cool if the configurations could be changed on the fly, from networktables or something. This would have to be done while disabled, but for things like fine-tuning PID or trying out different control schemes it could be useful to be able to change the settings so rapidly.

KyleRAnderson, 2018-11-18:
Another question, do the variables need to be constants? As in what's wrong with having them as methods? Is it just less easy to do `getCameraIp()` every time you want the camera's ip address?

KyleRAnderson, 2018-11-22:
**Team Decision**: We're going to use `static constants` for things which will never truly change, and then getters (and setters) for value which could potentially change.

greyingjay, 2018-11-27:
Guys, I'm impressed!

KyleRAnderson, 2018-12-11:
Together with #18 and #20 this has been implemented and tested. Meets all of above requirements.

closed 2018-12-11 by KyleRAnderson