FRC2706/2019-2706-Robot-Code #14: Differences between XBox 360 versus XBox One

greyingjay, 2018-11-27:
We are getting pretty good at using XBox 360 controllers and mappings for all of our robot code.

However, the XBox 360 controller is discontinued, and XBox One controllers are plentiful.

It would be useful to know how exactly these two map in the robot code. Do they behave identically? (I think the answer is no.) If not, can we create a mapping and/or subclass so we know how to work with them?

I think this is a good little project for a rookie student.

KyleRAnderson, 2018-11-27:
I've actually experimented a little with this before, I have a couple of notes from when I did it.

The Xbox 360 and Xbox One controllers both seem to work pretty much identically, except I've noticed issues with the Select/Back/View button ([see diagram](https://upload.wikimedia.org/wikipedia/commons/2/2c/360_controller.svg)) on both types of controllers, but I'm not sure if this has always been the case. 

Good idea to have another student take a look at it.

eandr127, 2019-01-26:
@Dimatsa's findings:

> From what I can tell, there is no difference between an Xbox 360 controller and a Xbox One controller [because they use the same input drivers, and all button mappings are the same on the Driver Station software]

closed 2019-01-26 by eandr127