FRC2706/2019-2706-Robot-Code #16: Write up testing procedures

KyleRAnderson, 2018-11-28:
Based off of @robertlucyshyn [comment](https://github.com/FRC2706/2019-2706-Robot-Code/issues/1#issuecomment-442247961) on #1, we think it would be a good idea for our testing procedures and/or methods to be documented (a "quick start guide" if you will) to aid with on-boarding rookies as well as to keep testing procedures consistent within the robot development team.

Would need to include:
- [ ] How we test for multiple hardware configurations (if we test this at all).
- [ ] When we test and how we do it so that everything that needs to be tested is tested.
- [ ] How we ensure that code merged to the master branch is tested (so maybe a tie in with how pull requests work and what we expect of code that is ready enough to have a pull request made).

greyingjay, 2018-11-28:
Please make sure you cover off how to get setup in whatever IDEs
people will be using. VS Code, since it is now the official IDE, and
if you guys want to use IntelliJ or whatever other tools, obviously
make sure people know how to get set up there too.

greyingjay, 2018-11-28:
(redacted, commented on wrong issue)

robertlucyshyn, 2018-11-28:
Yes, that would be great. To expand on this, one idea to consider is to develop a set of so called "regression tests" to check whether recent code changes have messed up the build, i.e. caused it to "regress". Regression can include breaking existing functionality or degrading performance (which can include things like slowing down processing or using up memory or other resources unexpectedly). Of course, this means that specific tests for these things must be written, and that comes down to what is practical and gives us a worthwhile return on the investment of time. Another thing to consider is that once such tests are up and running and procedures are written, they can be handed over to rookies to execute on future builds. This allows rookies to have hands on experience with the equipment and learn about testing, and frees up your own time so that you can put your experience to where it is most valuable. (Of course, I'm not suggest that rookies do only testing; I think they should all have some opportunity for development according to their ability.) As far as when to run such regression tests, one reasonable approach is after a merge to the latest build is done but while the changes can be easily backed out of if it is found that they have caused regression tests to fail. Since I have not personally been through a build season, I don't know how much of the preceding is practical given time and resource constraints, but it is something to think about.

eandr127, 2019-02-15:
I made a wiki page [here](/FRC2706/2019-2706-Robot-Code/wiki/Testing) that completes this issue.

closed 2019-02-15 by eandr127