FRC2706/2019-2706-Robot-Code #27: Continuous Integration Framework

KyleRAnderson, 2019-01-10:
It would be super useful to get some sort of continuous integration service up and running for the team to begin benefiting from automated testing. 

What needs to be researched:
- [ ] Continuous integration services for GitHub.
- [ ] Integration and Unit testing (and maybe other types of testing as well...)
- [ ] Testing frameworks to be benefited from.
   - See if there are any that are specific to FIRST robotics that might be useful.

KyleRAnderson, 2019-01-10:
I'm currently working on this in a [document](https://docs.google.com/document/d/1FISj_INc-XTBHHWTQTA4nwC3GKVuBPbtHgWYxuJJER4/edit?usp=sharing) in the team drive.

KyleRAnderson, 2019-01-11:
Decisions that have been made while doing research:
1. Use JUnit tests for unit testing. They're also used while doing JMockit tests as well.
1. Use Travis-CI connected to the team's GitHub account for automated testing.
1. Try to use JMockit for integration testing - super useful for mocking data, such as data from sensors.

All this has been recorded in the [document](https://docs.google.com/document/d/1FISj_INc-XTBHHWTQTA4nwC3GKVuBPbtHgWYxuJJER4/edit?usp=sharing) as well.

KyleRAnderson, 2019-01-13:
[First build passed on travis-ci](https://travis-ci.com/FRC2706/2019-2706-Robot-Code/jobs/170055044).

KyleRAnderson, 2019-01-13:
This article pretty much describes some of what needs to be done to get started testing: https://medium.com/@mglover/automated-java-tests-with-gradle-and-travis-ci-a6c8897231a2.

KyleRAnderson, 2019-01-24:
Added JMockit dependencies to the build file in dd9e24a8c1d3d1f6e91caf65ee4d7f1c6200adcf. This means we're ready to start making tests.

Tests can be run via the gradle task `check` (`./gradlew check` or `gradlew.bat check`) and once the branch is pushed travis-ci will run that same command to test the build.

greyingjay, 2019-01-24:
Cool. How do we know the results of Travis CI? Does it send emails?

KyleRAnderson, 2019-01-24:
So @greyingjay you'll notice in our README that there's now a [image] build status. This lets us know the state of the code in master.

You can click on this and you should be taken to the travis-ci dashboard for our repo (I'm not sure if it's just me who can see it. If you try it, let me know if it works). 

Otherwise, for each new branch created with the configuration file `.travis.yml` inside it, travis will run a build for the branch and email the person who committed the result of the build. After that it will only notify the person if the build fails.

Pull Requests have a dedicated section on them for the build status of the PR. I've set it up so that you can't merge PRs until the build passes.

greyingjay, 2019-01-24:
Aha! There it is. Very cool.

KyleRAnderson, 2019-01-27:
Another update.

We were having some trouble with what we thought was JMockit mocking on the CI server but it actually turned out to be that the WPIlib suite needs ubuntu 18 and above to run while the server (and most other ci servers) were using ubuntu 16 or below. 

This has lead us into using a docker container to run the builds on.

I set up an azure pipeline (another ci service) and @eandr127 helped me configure it to use the wpilib docker container. Tests are passing on that server now.

I'm still looking in to getting a docker working with travis-ci since it does still offer a couple of benefits that I like. I'm currently looking into a custom gradle build task for running the tests in docker, as this would allow us to run exactly what the server is running locally and to switch easily between ci services.

[This](https://bmuschko.com/blog/docker-integration-testing/) is the article I was looking at for this.

KyleRAnderson, 2019-01-28:
At this point I'm going to defer further investigation into getting docker working with travis since it's a pain and azure pipelines already works. 

Very soon (after I test the changes I made to the bling code, hopefully tonight), I'm going to merge the azure pipelines additions as well as my tests for the bling controller so others can begin to make tests as well.

These changes also include removing the travis configuration which will make travis stop running our builds.

KyleRAnderson, 2019-01-31:
Basic mocked testing and unit tests have been created and merged in #47.

We are now ready everyone to begin creating unit and integration tests for their own code.

closed 2019-01-31 by KyleRAnderson