FRC2706/2019-2706-Robot-Code #77: Button Assignments for DriverAssist

RachelLucyshyn, 2019-02-16:
RCVI needs to assign three buttons on driver controls to initiate 3 actions.
We want the top three buttons that are part of the four button group on the right-hand side. 

**Assignments:**
DriverAssist with vision (rocket): Right (B)
DriverAssist with vision (cargo): Left (X)
DriverAssist with laser rangefinder: Top middle (Y)

eandr127, 2019-02-16:
You can add these buttons into OI like this:

Config.java
```Java
// #### Fluid constants ####
...
public static final FluidConstant<String> NAME_OF_BUTTON = constant("button-name", XboxValue.XBOX_{letter}_BUTTON.NTString);
```

OI.java
```Java
public OI(Joystick driverStick, Joystick controlStick) {
...
    new FluidButton(driver/controlStick, Config.NAME_OF_BUTTON).whenPressed(new CommandToRun(...));
}
```

eandr127, 2019-03-07:
Has this issue been resolved?

KyleRAnderson, 2019-03-19:
The driver controls have been added, resolving this.

closed 2019-03-19 by KyleRAnderson