FRC2706/2019-2706-Robot-Code #79: Mechanisms

KyleRAnderson, 2019-02-18:
The robot will have mechanisms! We need code to run those mechanisms, which means new subsystems and new commands for certain actions.

KyleRAnderson, 2019-02-18:
@AbhiJ2706 has pretty much already completed all the mechanism code, however it needs testing as well as to get automated tests.

@AbhiJ2706 since you're going to be working with vision more in the next couple of weeks, do you mind if we continue with the work you've done on the mechanisms?

AbhiJ2706, 2019-02-18:
@KyleRAnderson automated tests are pretty much done. You can continue with the work, however, I suspect that vision will be done this Friday (as that is when they are comparing finished java v python) so I will be back with robot code as early as this coming Saturday so I will be able to pick it back up then.

greyingjay, 2019-02-18:
Please do as much physical testing as you can with the "stub"
mechanisms on the mezzanine (top layer) of Plyboy -- Controls put a
lot of effort into making those available so we can test items out for
real ASAP.

You can use the winch to test out the lift code. There should be an
encoder. Verify with controls that this all works. The CAN IDs on
Plyboy will end up being the same as the real robot.

You can use the standalone motor to test out the intake motor code.
Again, verify that this all works.

You can use the PCM to test out the pneumatics code. You don't have an
actual compressor on Plyboy but you can watch the LEDs on the PCM to
verify that solenoids would be told to open or close.

We need to define which PCM solenoid IDs correspond to which physical
function on our mechanism (intake arm up/down, hatch puncher in/out).
Come up with something, and make sure controls agrees to it and wires
it accordingly.

KyleRAnderson, 2019-02-20:
Intake subsystem is ready for testing on the real robot. Was having problems with the lack of a plunger solenoid on plyboy so waiting for practice bot.

greyingjay, 2019-02-21:
Plyboy will support pneumatic solenoid testing once we power up the PCM. I think all we need to do is provide power from the PDP (and a breaker).

KyleRAnderson, 2019-02-23:
While waiting for plyboy's PCM to be wired (@Ketchperiment was working on it) I'm working on the Lift subsystem fixing up.

KyleRAnderson, 2019-03-01:
See Erik's notes for how we've been implementing subsystems: https://docs.google.com/document/d/11xphTAxiSB7RdZs1SadKbBgYSWKwoQCAaZzwyg7_LUk/edit.

KyleRAnderson, 2019-03-02:
Mechanisms were tested tonight and the subsystems themselves work nicely. Most commands are good too. Need to work on lift smoothness (PID) and automatic things for operators.

PR #112 (2019-03-11): Mechanisms
Since we are running out of time and everyone else is waiting on mechanisms to get their stuff done, this is allowed to be merged without automated tests for now. Automated tests should come later.

Closes #109, closes #103, closes #107, closes #87, closes #86, closes #79.

## Summary of Changes
- **Added all the mechanisms.**
- Slight improvements to FluidButton so that Joysticks could be used with it.
- A lot of overall changes. Mostly additions.

## Testing Performed
**Environment**: All mechanism code has been tested on the practice bot (cosmobot).
closed 2019-03-11 by KyleRAnderson