FRC2706/2019-2706-Robot-Code #86: How will we determine the starting configuration of the intake arms?

KyleRAnderson, 2019-02-20:
One of the things that came up while I was working on the intake subsystem was the following question: how do we know what configuration the intake arms are starting in?

If we wanted to preload a ball, we would need the arms to be down. If we have a hatch, they would be up. This question can be divided into two question then:

1. Can the arms be down in the robot's starting configuration?
1. If the arms can be down, how will the robot know this?

greyingjay, 2019-02-21:
My understanding is that the arms must be up in the starting configuration. There is no ability to preload a ball.

Well, there may be theoretically a way, but it's silly. Maybe something we could do for a reveal video trick, like last year's Fidget Spinner Mode.

KyleRAnderson, 2019-03-02:
What we're currently thinking of doing now is the following:
1. If the robot starts with arms down and no cargo is in the arms (read from IR sensor), attempt to raise the arms on robot start.
1. If the robot starts with cargo in it (IR sensor), set the intake state to cargo and continue on.

PR #112 (2019-03-11): Mechanisms
Since we are running out of time and everyone else is waiting on mechanisms to get their stuff done, this is allowed to be merged without automated tests for now. Automated tests should come later.

Closes #109, closes #103, closes #107, closes #87, closes #86, closes #79.

## Summary of Changes
- **Added all the mechanisms.**
- Slight improvements to FluidButton so that Joysticks could be used with it.
- A lot of overall changes. Mostly additions.

## Testing Performed
**Environment**: All mechanism code has been tested on the practice bot (cosmobot).
closed 2019-03-11 by KyleRAnderson