FRC2706/2019-2706-Robot-Code #92: Bling Patterns

KyleRAnderson, 2019-02-23:
Currently the robot's bling subsystem is ready for patterns, and has only one: nothing.

It would be great to get some patterns going, however there are a couple of things that need to be done as part of this.

- [ ] Intake and Lift subsystem have to be merged to master, or to their own master branch.
- [ ] Need feedback from drive team on what patterns they would like to see.
- [ ] Need information from controls about where the LED strips are mounted and how big they are (to determine how useful they're going to be).

KyleRAnderson, 2019-03-13:
No bling hardware-wise on competition bot, no need to create bling patterns this year.

closed 2019-03-13 by KyleRAnderson

greyingjay, 2019-03-13:
.... yet! The hope is to get it on for North Bay. The bling students
worked hard on patterns, at least they did earlier in the build
season.