FRC2706/2019-2706-Robot-Code #106: Dashboard set up

KyleRAnderson, 2019-03-04:
One of the large requirements for this year (given the sandstorm period) is some sort of nice dashboard set up. It would need the following features:

- [ ] Shows lift motor current/torque on the lift motor.
- [ ] Two resizable camera feeds for the two cameras on the robot.

We can probably accomplish this using shuffleboard.

I'm going to be very busy with mechanisms and I think @eandr127 is going to be doing auto modes so @carmen1234 or @Zinka010 if either of you are willing to do this that would be great.

greyingjay, 2019-03-04:
And because we all love extra challenges:

Might be useful to think about a way to warn the user if the lift is being over stressed. Like maybe it gets backlit in red and flashes when the motor current is above some value X. Just an idea.

Second, my understanding is that one of the camera views should come directly from a USB connection on the RoboRio, and another one should come from a camera server stream from the vision Pi. It would be desirable to have both show up on the dashboard. For today, they were having difficulty getting the Pi talking on the network so we connected both cameras to the USB ports on the Rio and were able to see two streaming images for drive practice. But, ideally (since we want to see vision working!) this is not how it will remain.

KyleRAnderson, 2019-03-13:
This has been accomplished with Shuffleboard. We simply need to save the config and it'll open with the cameras that we want.

closed 2019-03-13 by KyleRAnderson