FRC2706/2019-2706-Robot-Code #127: A mechanism to override the ring lights on for testing

greyingjay, 2019-03-18:
RCVI team would like a way to tell the ring lights to turn on and stay on - for example to record the camera images during a full practice match at North Bay in order to get sample images for testing.

A potential way to do this is to short the ring light power wires across the relay so they are wired to always-on.

Can we do it in software with a dashboard selector? It doesn't need to be checked every teleop tick, it could just be a setting that gets checked when the robot is initialized and then sends a single call to your ringlight.enable() method.

KyleRAnderson, 2019-03-18:
Yep, we could do this, though we can't control the ring light while the robot is disabled. It will always be off.

Would this override all calls to turn the ring light off then as well?

greyingjay, 2019-03-18:
My understanding is they just wanted a quick way to be able to turn the light on without having to worry about commands (for example if the vision integration isn't quite ready or turned on). I will ask if they need this override to really override all existing ring light commands or if it's OK that subsequent commands could turn the ring light off again.

In a related question, the RCVI mentors mentioned that they think there may be a bug in the commands that cause the ring light to get turned off again when it shouldn't be. Not sure if this is still true or reproducible. We should investigate as part of this work.

robertlucyshyn, 2019-03-18:
I understand that the need to permanently turn on the ring light is just for the vision people to collect data during a practice match. I don't think it was to be used in competition.
I think the bug referred to below (it is exists) is not related to the turning off of the ring light itself but rather the tapeDetected flag in the network table being set to true too early, and then the command progressing through its normal course of computing a trajectory, commanding the motion control system, and then turning off the ring light. I'm sure we'll have a much better understanding once we get to run some tests on the practice robot which we have not yet done at all---except for some tests in the pit with the robot elevated above the ground with its wheels spinning.
All the above being said, it may be worthwhile to have commands that turn the ring light off and on, rather than the current command that toggles its state. That way, the command can still be used when the ring light is permanently turned on without messing things up.
Robert

kevlam2706, 2019-03-25:
I think this issue is resolved?

kevlam2706, 2019-04-02:
I am closing this issue since it now appears to be resolved as we have decided to keep the ring lights on all the time.

closed 2019-04-02 by kevlam2706