FRC2706/2019-2706-Robot-Code #128: Check for encoder presence on startup

greyingjay, 2019-03-18:
The autonomous failure at Ryerson was due to the encoders being wired to different Talons on the practice and competition robots. However, a similar failure could occur for any number of reasons including encoders falling off (which happened in Stronghold) or damaged (Steamworks) and the same problem would occur if the encoder wires were cut or broken or if the Talon's Gadgeteer port were damaged.

Check out what 254 does: https://github.com/Team254/FRC-2018-Public/blob/master/src/main/java/com/team254/frc2018/subsystems/Drive.java

`final ErrorCode sensorPresent = talon.configSelectedFeedbackSensor(FeedbackDevice
                .CTRE_MagEncoder_Relative, 0, 100); //primary closed-loop, 100 ms timeout
        if (sensorPresent != ErrorCode.OK) {
            DriverStation.reportError("Could not detect " + (left ? "left" : "right") + " encoder: " + sensorPresent, false);
        }`

Please implement a similar check on our drivetrain and lift encoders. Test by swapping the talon IDs temporarily and verify that it detects that the encoder isn't there.

We might even want to disallow autonomous driving commands if we think we've lost an encoder.

greyingjay, 2019-03-18:
I butchered the code block formatting. See line 93 in the source I linked to.

greyingjay, 2019-03-21:
Here's another great example from the Poofs. Check out the constructor:

https://github.com/Team254/FRC-2018-Public/blob/master/src/main/java/com/team254/frc2018/subsystems/Elevator.java

eandr127, 2019-03-23:
Completed by #137

closed 2019-03-23 by eandr127