FRC2706/2019-2706-Robot-Code #129: Prepare for a climbing mechanism

greyingjay, 2019-03-19:
Design & Fab team is going to meet this week to design a climber that can be added to the robot. The schedule is tight, and they will need software to test by the weekend.

Let's try to stay ahead of the curve by preparing the software to support the climber. There are many things we obviously don't know yet, but let's make some "worst case" assumptions, create the mechanism code, and then adjust when we know more. 

Let's assume:

- it will be driven by a single motor controlled by a Talon SRX (so we'll need to assign it a CAN ID)
- it will have both a forward and reverse limit switch to stop travel (no idea if it really will, but code is faster to delete than add)
- it will require current-limiting
- it will require driver or operator controls to deploy and retract (so assign some button mappings - make sure they're unlikely to get accidentally triggered during normal teleop stuff)
- assume we would drive the motor at some fixed speed (slow-ish) by manual control, and assume we will need a way to adjust this fixed speed quickly

KyleRAnderson, 2019-03-19:
Likely going to be mapped to X and B on operator controller after I move the override lift stuff to the left stick.

KyleRAnderson, 2019-03-20:
Basic climber subsystem and commands have been completed. Just need testing.

greyingjay, 2019-03-21:
Thanks Kyle. Some additional requirements, based on tonight's discussion with the Design & Fab people. The basic behaviour of the proposed (and still theoretical) climber is a bar attached to a motor that would rotate downward, lifting up the front end of the robot to the top lip of hab level 3. Then two pneumatic pistons would fire to lift up the back of the robot, _while_ the bar is still rotating and pushing the robot forward onto the hab platform. The pistons would retract, but the bar would remain hanging downwards, but not touching the level 1 hab surface.

Based on this:

- Let's assume CAN ID 7 as the next logical progression.
- Let's assume there's an encoder on the motor. This might negate the need for a limit switch if we set a soft limit at 0.
- The two pistons would be on PCM (4,5) and (6,7).
- Your two button mappings are now: one button to lower the bar, another button to extend/retract the pistons.
- We might need to come up with an automatic sequence where the pistons extend automatically when the bar reaches some position X on the encoders.

Go go go!

KyleRAnderson, 2019-03-22:
I'll add an additional note: might make sense to disable the compressor while climbing to maximize power available. Would that make sense @kevlam2706?

kevlam2706, 2019-03-22:
Sure. Wouldn’t hurt. Just want to make sure this is tested so we don’t accidentally impact regular teleop game play.

KyleRAnderson, 2019-03-25:
Some more things that I need to do:

- [ ] Remove compressor turning off
- [ ] Raise intake arms (if they are not already up)
- [ ] Add voltage ramp so motor doesn't go too fast initially.

kevlam2706, 2019-03-25:
Yes, we'll remove the compressor disable because those huge cylinders might require the compressor to keep running to pump the air. This might actually interfere with the idea of a timed sequence. We'll need to test this to make sure that a timed sequence won't fail and tip the robot over if the pistons are moving more slowly because of lack of air. There are possible ways to mitigate this including plumbing a few air tanks with a check valve so that they will only feed the climb pistons.

kevlam2706, 2019-03-25:
So to summarize:

- [ ] One motor driven by a Talon SRX, CAN ID 7. 
- [ ] Has an encoder.
- [ ] Reverse limit switch to prevent backdriving into the frame. 
- [ ] Set a current limit
- [ ] Set a voltage ramp that's somewhat slow
- [ ] Soft limit on the other side. NO physical limit switch. 
- [ ] Constant to determine the soft limit encoder value.
- [ ] Two pistons. PCM ID's 4,5 and 6,7.
- [ ] Even PCM ID's (4 and 6) to deploy pistons.
- [ ] Odd PCM ID's (5 and 7) to retract.
- [ ] When the climb sequence is engaged, run the motor at 100% output
- [ ] When the bar reaches the climb threshold, fire the pistons
- [ ] Constant to determine the climb threshold value
- [ ] Do NOT run the motor backwards, at least not for level 3 climb. There is no "retract" for the bar.
- [ ] Retract the pistons at the end of the climb sequence
- [ ] Probably want to set all this to run on a "whileHeld" button trigger so you can cancel by letting go in case the climb isn't going well? When the button is let go the command could end by stopping the motor and retracting the pistons.

kevlam2706, 2019-03-25:
One potential other use: We might be able to use the mechanisms to also do a level 2 climb. One way to do it might be to run the arm to a smaller angle (enough to clear the level 2 platform) and then push out the pistons, let them go out half-way or so, then pull them back in (i.e. pull them in before they have fully extended).  We MIGHT need to be able to retract the arm back in and drive forward after that. I don't think we need to make any changes right now, but keep this in mind as a possible secondary use.

kevlam2706, 2019-03-26:
Good efforts yesterday. We will revisit this but it will likely not be until after North Bay. It will be in a new gearbox which means new encoder values all around.

PR #162 (2019-04-15): Changes from FOPC
## Summary of Changes
- Pneumatics wrapper class
- Climber subsystem and auto-climb command
- Change FMS connected to wait for match data
- Lift setpoint adjustment
Closes #129 and #159 

## Testing Performed
**Environment**: Competition bot
closed 2019-04-15 by eandr127