FRC2706/2019-2706-Robot-Code #131: Tune Lift PIDs

eandr127, 2019-03-19:
Right not the PIDs aren't tuned properly.

- Separate PIDs for position (holding) and velocity (and motion magic) control
- Follow PID tuning guide
- Ensure that the new PIDs work as well or better than the previous ones

kevlam2706, 2019-03-25:
I agree this should be done.

Given all of the moving parts everywhere else right now (bling? 3rd camera? vision assists? climber integration? autonomous?) I think we should lower the priority on this until after North Bay.

I don't think the lift speed is the thing that is keeping us from doing more cycles - right now our issues are with lining up, deploying, etc.

kevlam2706, 2019-04-02:
We did end up tuning the P value on the lift to increase the repeatability of the loading station preset. It currently, consistently, underruns the target value (e.g. tell it 2000 ticks and it consistently reaches 1700).

It would be ideal to have precision _and_ accuracy, but precision alone is sufficient for now as we can just adjust the presets as needed to achieve the actual positions we need even if the two aren't 100% the same. If we do improve the tuning, and it has the result of improving the underruns, we will need to recalibrate all the lift setpoints.

PR #169 (2019-05-04): Tune lift pids
## Summary of Changes
- Lift PIDs tuned
- Closes #131.
- Add lift feed-forward.
- Add a specific config slot for motion magic vs hold lift position.

## Testing Performed
**Environment**: Practice bot
Tuned the PIDs and made sure the changes didn't break things.
closed 2019-05-04 by KyleRAnderson