FRC2706/2019-2706-Robot-Code #136: Put LIDAR readings onto networktables

kevlam2706, 2019-03-21:
This might be for @robertlucyshyn and @RachelLucyshyn. Get the LIDAR reading and publish it to a networktable value (and possibly a smartdashboard value) for driver monitoring. The vision Pi could use the distance value to annotate the images it saves. This could be a good sanity check of the vision distance calibration.

KyleRAnderson, 2019-04-02:
One of myself or @Zinka010 can probably deal with this one.

robertlucyshyn, 2019-04-02:
The lidar code is not yet on master but is on the rcvi_laser branch. I had a question before North Bay on writing a regression test for it. 
Robert

Zinka010, 2019-04-03:
Would this be a FluidConstant that can be changed or simply a value that could be easily read on network tables?

KyleRAnderson, 2019-04-03:
@Zinka010 it should just be something put to SmartDashboard. Every tick you would simply have to run
```java
SmartDashboard.putNumber(someNumber);
```
with `someNumber` being the distance that our lidar code determines in feet.