FRC2706/2019-2706-Robot-Code #151: Tune Pathfinder

kevlam2706, 2019-04-02:
The night before we left for North Bay we were looking at tuning Pathfinder to help the robot drive arbitrary paths with good accuracy. We had at one point found that it seems to drive curves to the right more accurately than curves to the left. This in theory should not be the case, since even if the drivetrain is asymmetrically powered, the fact that we're tracking using gyro and encoders should automatically compensate.

There was also a recurring bug where the Pathfinder command seems to drive a previously-remembered trajectory.

Debug these issues. The goal is to drive any given trajectory accurately, either toward the left or right. If it helps, do so at a medium speed before trying to speed things up. Medium speed but consistent is better than high speed but unpredictable.

KyleRAnderson, 2019-04-02:
@eandr127 mind tackling this one?

robertlucyshyn, 2019-04-02:
I just want to mention that I ran a couple of Pathfinder trajectory following tests at Mission Control Monday night on the practice bot with the North Bay code and the results we pretty good. These tests were decoupled from vision. I had two trajectories: 1) 8 ft forward, 3.25 ft to the right; 2) 8 ft forward, 3.25 ft to the left. Both ended up almost dead on in the forward direction and only a couple of inches off in the side direction. Trajectory following for shorter distances is not quite as good when the robot has to do a sharper turn but that might be a fundamental limit of the mechanical device.

eandr127, 2019-04-09:
Closed by #160

closed 2019-04-09 by eandr127