FRC2706/2019-2706-Robot-Code #156: Motion Profile for Climber

eandr127, 2019-04-06:
It looks like the climber arm will need to run at certain velocities at different points during the climb process.

- Generate curve
  - Different target velocities for different stages of the climb process
- Follow curve (two possibilities)
  - Motion profile
    - Save trajectory to file and run as Talon motion profile
    - Better control, but may be harder to implement
  - Voltage control
    - Motor voltage as function of either time, gyro pitch or encoder ticks
    - Less control, very easy to write and integrate with sensors

eandr127, 2019-04-06:
@robertlucyshyn, from the email, it looks like you will be working on generating the curve. Were you also going to do the following part of it as well?

robertlucyshyn, 2019-04-06:
I’m not sure how to do the following part. Part of how depends on whether we want the talon to follow an encoder angle or motor speed. When I spoke we Brian, we were talking about motor speed but I realized later that encoder angle might be much easier to tune. If I give you either a time history of encoder angle or motor speed, do you know how to command the talon to follow it?

eandr127, 2019-04-06:
I wrote some code [here](/FRC2706/2019-2706-Robot-Code/tree/mp-climber) that we might be able to use to follow the velocities. What we would need is a list of either gyro or encoder position readings, as well as their velocities (Either encoder ticks per second or pitch degrees per second). We'd also need the time in milliseconds between data points.

The other way we could do it is have a map of encoder/gyro position to a voltage and use the entry with the closest reading to set the output.

robertlucyshyn, 2019-04-06:
I wrote a motion profile generator in Matlab in the mp-climber branch in src\main\resources\mp_generator_matlab that generates a profile as a series of smoothed steps. It is very general: you can specify any number of steps and for each step you specify 1) the value; 2) the start time; and 3) the end time. In between each step, there is a smooth transition. It also computes its derivative. We can use this if we want to specify motor speed as an input to the talons or if we want to specify encoder angle to the talons (or any other quantity we want to specify to the talons). If the input is an encoder angle, the derivative represents the encoder angular speed.

Here is an example if we specify motor speed:

[image]

Here is an example if we specify encoder angle:

[image]

For this last graph, here is the encoder angular speed:

[image]

eandr127, 2019-04-14:
Closed since we didn't use a climber motor on the real robot

closed 2019-04-14 by eandr127